UbiquityRobotics/magni_robot #14: Update for Magni V5

rohbotics, 2017-01-28:


rohbotics, 2017-08-23:
Definition of Done:

The URDF reflects the actual physical form of Magni V5 (Blue ones).

davecrawley, 2017-08-23:
Note - that the new Magni V5 (blue ones) have some slight physical differences to the previous iteration. I have step and solidworks models of the new version. Let me know how I get these in to the URDF.

rohbotics, 2017-12-27:
@davecrawley Are we still holding on the animators new 3d models?

davecrawley, 2018-01-27:
Camera Positions
Sonar Positions
Wheel positions
3-D sensor positions
Height of center point axles off ground

jim-v, 2018-02-25:
It would be good if we can flip the image so that it is upright.  This affects the camera orientation, and needs to be tested on the actual hardware.

rohbotics, 2018-06-04:
solved with #56 i believe

closed 2018-06-04 by rohbotics