UbiquityRobotics/magni_robot #33: Assemble High-Level List of Packages

rohbotics, 2017-07-29:
Should go in the README

jim-v, 2017-08-05:
Here is the list of packages

`ros-kinetic-desktop-full
ros-kinetic-magni_bringup
ros-kinetic-magni_description 
ros-kinetic-magni_robot
ros-kinetic-magni_teleop
ros-kinetic-magni_demos
ros-kinetic-magni_nav`

waynegramlich, 2017-08-05:
All:

Do we really want to install ros-kinetic-desktop-full on the Pi?  I want it preinstalled
in the VirtualBox image, but I am skeptical that it should be preinstalled on any of the
robot platforms.

My $.02,

-Wayne

rohbotics, 2017-08-05:
@waynegramlich This is for the virtual box.

Also here is the apt get command
```
sudo apt-get install \
ros-kinetic-desktop-full \
ros-kinetic-magni-bringup \
ros-kinetic-magni-description \
ros-kinetic-magni-robot \
ros-kinetic-magni-teleop \
ros-kinetic-magni-demos \
ros-kinetic-magni-nav
```

closed 2017-08-12 by davecrawley