UbiquityRobotics/magni_robot #59: Robot starts, but doesn't stop, driving in various situations (Regression)

davecrawley, 2018-06-11:
We've found that in the new image - certain behaviors are non functional. In particular when we give it a voice command to "go forward 10cm" it starts driving forward but never stops. We've also noticed that if we set a waypoint and tell it to go to it - it will start driving but never stop.

jim-v, 2018-06-11:
Suggest using `rostopic` to check that the odometry is working and that `move_basic` is subscribed to the correct topic.  A second suggestion is to have `move_basic` abort goals if odometry does not indicate progress towards them.

jim-v, 2018-06-11:
Please disregard my previous comment.  My suggestion is to view use rviz to determine if the robot is moving in the expected direction.  If it is moving backwards with respect to the map frame in rviz, then the camera is not mounted consistently with the way it is specified in the URDF.

closed 2018-07-22 by rohbotics