UbiquityRobotics/magni_robot #73: Robot drives at high speed upon loss of WiFi

matjazp, 2019-02-19:
When the robot looses wifi the robot apparently drives off at high speed for unexplained reasons. This occurred during a hackathon with multiple users logging in to the robot at the same time.

davecrawley, 2019-02-21:
Steps to make it occur
- Run a program any program
- After loss of WiFi
- Stop for 2 minutes 
- Robot will drive off at high speed!

rohbotics, 2019-02-22:
Can you provide more details on exactly what programs you tried?

Are you using move_basic, or cmd_vel only? What else is running?

davecrawley, 2019-02-23:
- Issue occurred when running either fiducial follow or Teleop
- Robot operating in AP mode
- Loss of Wifi induced by moving the robot away from laptop
- Only seemed to be on certain (perhaps a single example) RPI

brunolfm, 2019-02-27:
On my early tests this problem have happened to me too.. unfortunately i didnt have the bag file recording!

This was what i was running at that time: 

roslaunch pi_sonar pi_sonar.launch
roslaunch magni_nav move_basic.launch
roslaunch magni_nav aruco.launch
roslaunch magni_viz view_nav.launch

https://youtu.be/hNVPIjhQjwc