UbiquityRobotics/magni_robot #78: Uncontrolled movement after killing and restarting motor node

davecrawley, 2019-03-05:
Start magni motor node
Publish a twist message periodically on /cmd_vel (e.g., 0.1m/s forward, 30Hz)
Observe Magni moves forward as expected
Stop motor node
Observe Magni stops as expected
Stop publishing messages on /cmd_vel
Start magni motor_node
Observe Magni is moving at max velocity for a short period of time seemingly proportional to the original velocity.
This looks like the firmware’s integral is winding up when it is in an unconnected state.
Since we regard this kind of uncontrolled movement as a security issue, we would greatly appreciate help with it.`

hendrik-skubch, 2019-03-05:
Dear Dave,
here are some additional details:
- We observed this on firmware version 30.
- For reproduction we started the magni-base via roslaunch and terminated it with Ctrl+C.

mjstn, 2019-03-06:
Thanks for verification of this on rev 30.  Many features went into firmware version 32 and I was concerned this may be new but it is pre-existing.  Today this was duplicated on firmware version 29 which was our first official production run firmware.

mjstn, 2019-03-07:
Observations:
- Run joystick very slow prior to stop and the amount of 'catch up' is far less so speed of joystick is involved
- Wait a long time before the start and the amount of wheel movement is much greater.  
- Full removal of power to motor controller at some point after the stop and power restored before the start of magni service leads to this issue not happening.  (The removal of power to just the motor controller requires a custom hardware setup)

From the above we can then say this is certainly a firmware issue.

mjstn, 2019-03-07:
This problem is understood and a fix is being tested now.  We must regression test the firmware and then schedule it into a firmware release.  All this will take some time as a key member of our team is out for a few weeks we cannot form a firmware release easily just now.  Thank you for finding this issue.

bhuvanchandra, 2019-03-15:
@mjstn May I know the status of the fix ?

rohbotics, 2019-03-18:
We are working on the v32 firmware release, which incorporates a number of bug fixes. We are aiming for early next week for limited public availability, and if no critical errors are found we will push it out more widely.

bhuvanchandra, 2019-03-29:
@rohbotics @mjstn, v32 firmware released ?

bhuvanchandra, 2019-04-01:
@mjstn, @rohbotics,
After upgrading the firmware to v32 on Magni the uncontrolled movement issue seems to be solved. But there are continuous DESERIALIZATION ERROR (-5) messages in logs.

Logs:
> $ rosrun ubiquity_motor motor_node 
[ INFO] [1554085793.006914315]: Firmware version 32
[ INFO] [1554085793.007143180]: Firmware version 32
[ WARN] [1554085793.106713952]: Setting P to 5000
[ WARN] [1554085793.156718648]: Setting I to 0
[ WARN] [1554085793.206610491]: Setting D to 0
[ WARN] [1554085793.256563614]: Setting Denominator to 1000
[ WARN] [1554085793.306567125]: Setting D window to 10
[ INFO] [1554085793.309010333]: Firmware version 32
[ WARN] [1554085793.309320463]: Resetting controller due to time jump 0.252274 seconds
[ERROR] [1554085795.232986648]: DESERIALIZATION ERROR! - 5
[ERROR] [1554085795.342393461]: DESERIALIZATION ERROR! - 5
[ERROR] [1554085795.483972679]: DESERIALIZATION ERROR! - 5
[ERROR] [1554085795.593554499]: DESERIALIZATION ERROR! - 5
... 

Also one more observation regarding the emergency stop, the issue seems to be still persistent with v32, I was under an impression that this fix also solves the emergency stop issue.

mjstn, 2019-04-01:
Firmware v32 is one of 3 parts of what is required to fix the ESTOP issue properly.
1) Motor controller board of version 5.0 or later with hardware support to read ESTOP switch state
2) Firmware that can detect the state of the ESTOP switch
3) Host ROS side code that tells the firmware it is ok to 'believe' the ESTOP switch.

The 3rd item, host side code, is not released yet so the full ESTOP fix cannot be taken advantage of yet.

As far as the DESERIALIZATION ERROR it will be investigated starting tonight.  They seem to be every 110msec apart from the 4 you show.

We have a way to go back to the prior old stable release for the firmware which at this time is version 30.
You have the power then to backup to the prior firmware if desired.
I am going to get that documented on the 'Learn' page that shows firmware update but have been told it is use of entering of  'oldstable'  for the firmware version.   I am not the 'expert' in the update utility.

mjstn, 2019-04-01:
Can you tell me a little more on the conditions of your system for getting the DESERIALIZATION error?
- Are you doing modifications to ubiquity_motor node for your own needs?
- Are you using any code in your system that writes to /dev/ttyAMA0
- Are you using a different serial port than the Raspberry Pi default serial port ttyAMA0

It looks like you are not using the normal startup service and are running the motor_node using manual command line rosrun.   Are you very sure everything else is running but the motor node is NOT yet running when you do this?   Please attach the sequence of commands you use from a restart of the RaspberryPi or other cpu.

bhuvanchandra, 2019-04-01:
> Can you tell me a little more on the conditions of your system for getting the DESERIALIZATION error?
>
> * Are you doing modifications to ubiquity_motor node for your own needs?
> * Are you using any [...]

Our setup doesn't have a Raspberry Pi. We interfaced the Magni motor controller serial port via USB-UART(FTDI) to Intel NUC. The Magni_motor node is installed on the Intel NUC. (Note: No changes done to the motor node code). One minor change we did is to the python script. Since we use different /dev/tty* instance, we updated the firmware updater python script to use appropriate /dev/tty* instance. Flashing works without any issues.

[Launch file](https://pastebin.com/JCsq3zqs) and [Config file](https://pastebin.com/yhnVpzmS) which we use.

Note: The DESERIALIZATION errors are only observed after updating the firmware to v32

bhuvanchandra, 2019-04-01:
> Firmware v32 is one of 3 parts of what is required to fix the ESTOP issue properly.
>
> 1. Motor controller board of version 5.0 or later with hardware support to read ESTOP switch state
> 2. Firmware [...]

Feature for sensing the ESTOP switch is available in v32 firmware ?

> As far as the DESERIALIZATION ERROR it will be investigated starting tonight. They seem to be every 110msec apart from the 4 you show.
>
> We have a way to go back to the prior old stable release for [...]

I do have the steps, which earlier were provided by you. But we are testing v32 firmware, to make sure the issues were fixed. All the issues reported were critical, and we cannot confidently use the robots with issues fixed.

closed 2019-04-07 by rohbotics