UbiquityRobotics/magni_robot #83: BMS

pyhuang2011, 2019-03-13:
Have a BMS according to regulation(should let the system know over current/over temperature...)
at least know is it charging.

davecrawley, 2019-03-13:
Please point to the regulation you are referring to or give a comprehensive description. BMS for over temperature are necessary for LiIon batteries due to their tendency to catch fire. Temperature is much less significant for Pb Acid

pyhuang2011, 2019-03-14:
there are 2 parts,
one is a general description in regulation when you have issues related to safety you should stop the machine, I think over current and over temperature is. - need to know and record for diagnose.
the other is we want the robot to know it is charging so it can disable Movement，unless human remove the charging plug

davecrawley, 2019-03-14:
Well we have temperature measurement in the hardware which we can probably expose - so let's do that - it will take some time. Current measurement is in there too - we should expose that too.

pyhuang2011, 2019-03-14:
so if you define discharge current is -xxx A, then charge current is + xxx A

davecrawley, 2019-03-14:
That's not quite how the current sensing works. It detects current being
used by the robot systems not net current in to the battery. In charging,
the robot will still show power consumption even if some of this current
comes from the charger rather than the batteries.

As such

+ power consumption and +dV indicates charging
+ power consumption and -dV indicates disconnected from the charger

The other thing that's important is that the charger boosts the voltage
quite a bit - again the is Pb-acid not LiIon so the voltage jump is
significant and is probably the easiest way to detect the event of being
plugged in. In fact you could write the voltage to non-volatile memory
every say 10 seconds and compare against that. That way even if the robot
was shut-down accidentally during charge up the system would be able to
tell.

David

David