UbiquityRobotics/magni_robot #97: Fixed Frame [odom] does not exist

asergeenko, 2019-07-28:
I got this error message when running "roslaunch magni_gazebo empty_world.launch":
Fixed Frame [odom] does not exist

Log says:
[DEBUG] [1564057088.520718745, 59.439000000]: Error transforming from frame 'sonar_0' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.522959976, 59.439000000]: Error transforming from frame 'sonar_1' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.524712364, 59.439000000]: Error transforming from frame 'sonar_2' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.525548225, 59.439000000]: Error transforming from frame 'sonar_3' to frame 'odom': "odom" passed to lookupTransform
[DEBUG] [1564057088.534252142, 59.441000000]: Error transforming from frame 'sonar_4' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.537619732, 59.441000000]: Error transforming from frame 'base_footprint' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.538769269, 59.441000000]: Error transforming frame 'base_footprint' to frame 'odom'
[DEBUG] [1564057088.541706730, 59.441000000]: Error transforming from frame 'base_link' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.543376257, 59.441000000]: Error transforming frame 'base_link' to frame 'odom'
[DEBUG] [1564057088.547622199, 59.442000000]: Error transforming from frame 'left_caster_wheel' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.549519024, 59.442000000]: Error transforming frame 'left_caster_wheel' to frame 'odom'
[DEBUG] [1564057088.550640581, 59.442000000]: Error transforming from frame 'left_wheel' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.554245715, 59.442000000]: Error transforming frame 'left_wheel' to frame 'odom'
[DEBUG] [1564057088.555186458, 59.442000000]: Error transforming from frame 'raspicam' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.556215378, 59.442000000]: Incoming queue was full for topic "/clock". Discarded oldest message (current queue size [0])
[DEBUG] [1564057088.556908288, 59.444000000]: Error transforming frame 'raspicam' to frame 'odom'
[DEBUG] [1564057088.561169137, 59.444000000]: Error transforming from frame 'raspicam_sensor' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.564044565, 59.444000000]: Error transforming frame 'raspicam_sensor' to frame 'odom'
[DEBUG] [1564057088.564364803, 59.444000000]: Error transforming from frame 'right_caster_wheel' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.568688934, 59.444000000]: Error transforming frame 'right_caster_wheel' to frame 'odom'
[DEBUG] [1564057088.571341114, 59.444000000]: Error transforming from frame 'right_wheel' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist. 
[DEBUG] [1564057088.571845636, 59.444000000]: Error transforming frame 'right_wheel' to frame 'odom'
[DEBUG] [1564057088.573528812, 59.444000000]: Error transforming from frame 'sonar_0' to frame 'odom': "odom" passed to lookupTransform argument target_frame does not exist.

rohbotics, 2019-07-28:
Can you run `rosrun tf2_tools view_frames.py` and send me the resulting frames.pdf? This shows a diagram of the entire tf frame tree.

asergeenko, 2019-07-28:
[frames.pdf](https://github.com/UbiquityRobotics/magni_robot/files/3439324/frames.pdf)

asergeenko, 2019-08-08:
I forgot to install the dependencies:
`rosdep install --from-paths src --ignore-src --rosdistro=kinetic --skip-keys="libraspberrypi0 libraspberrypi-dev" -y`

closed 2019-08-08 by asergeenko