UbiquityRobotics/magni_robot #108: Add "Disable Time Sync" option to robot.yaml

cnboonhan94, 2020-01-23:
I refer to this issue on [Discourse](https://forum.ubiquityrobotics.com/t/certain-topics-take-a-long-time-before-publishing/439/11). 
In summary, for scenarios where Magni is connected to WiFi without an internet connection, the Magni waits for the full 20 iterations for `chrony` to sync, as specified [here](https://github.com/UbiquityRobotics/magni_robot/blob/indigo-devel/magni_bringup/scripts/launch_core.py#L67). This delays the launching of the motor node significantly. 
It might be worth looking into an option of disabling clock sync in such scenarios as a parameter in `robot.yaml`. For coordinated multi-robot scenarios in this environment, however, clock sync is necessary and a local time server would probably be required.

rohbotics, 2020-01-23:
@cnboonhan94 Thanks for opening the issue. It won't be the highest priority item, but we will try to address it soon.

cnboonhan94, 2020-01-23:
@rohbotics its no issue! Thank you for your swift assistance.

rohbotics, 2020-02-01:
Hi @cnboonhan94 

I made a PR for this issue https://github.com/UbiquityRobotics/magni_robot/pull/110

It allows you to add `force_time_sync: False` into your robot.yaml and disable all of the chrony time sync stuff. Let me know if this addresses your issue, or if there is something else that would be better.

cnboonhan94, 2020-02-13:
apologies for the delay, i have not been able to look at this. I will try this out on the Magni as soon as possible, hopefully tomorrow. Thank you so much!

closed 2020-02-17 by rohbotics