UbiquityRobotics/magni_robot #120: Resource not found: Raspicam Node

maryamnassar, 2020-07-02:
Hello, I am trying to run the Magni on Ubuntu 16.04 and Gazebo 7. I have followed the directions outlined here: https://learn.ubiquityrobotics.com/simulation and am able to spawn the Magni in an empty world. However when I attempt to run a demo and input the command roslaunch magni_demos simple_navigation.launch I get the following error:
[image]
How can I fix this and launch the navigation demo?

tp4348, 2020-09-17:
@maryamnassar please give a bit of more insight what are you struggling with.

MoffKalast, 2020-09-17:
I'm pretty sure he means that the fiducial slam/follow launch file also starts up raspicam_node. There should probably be separate launch files for gazebo and real robots.

JanezCim, 2020-10-08:
@maryamnassar or @rohbotics can you tell me how to reproduce this problem? Running `roslaunch magni_gazebo fiducial_world.launch` I do not encounter any problem. I am doing this on branch `indigo-devel`

closed 2022-04-20 by JanezCim