UbiquityRobotics/magni_robot #126: in new way for pid params to be in robot.yaml if they are missing some params are mixed up

mjstn, 2020-08-03:
Using  a clone of magni_robot it is expected that pid parameters are in robot.yaml file as well as many more things.

If robot.yaml is still the old one in /etc/ubiquity/robot.yaml then  derivative and denominator are mixed up.

- The default for  pid_derivative will end up being  pid_denominator setting at runtime taken from default_conf in launch_core.py

- The default for pid_denominator at runtime will end up being controller_pid_denominator default from core.launch

I think proportional and integral defaults come from one or the other but both were same for me so I don't know if that is issue as well.

mjstn, 2020-08-03:
I am suspecting this is some sort of side effect of the names being similar and somewhere one is used in place of other but I did not locate it myself in my browsing.  It may also be due to the clever use of a large container to hold default_conf and pass into roslaunch.main through sys.argv.  Maybe an ordering thing or a typo thing for similar names.

rohbotics, 2020-08-03:
I found it, making a PR now. https://github.com/UbiquityRobotics/magni_robot/blob/indigo-devel/magni_bringup/scripts/launch_core.py#L135

closed 2020-08-07 by rohbotics