UbiquityRobotics/magni_robot #131: Drive magni with Joystick

tp4348, 2020-09-21:


rohbotics, 2020-09-21:
Could you provide more detail on this issue? This could be a couple different things, is this for the BLE joystick?

tp4348, 2020-09-21:
This is Logitech Gamepad F710 joystick that was shipped to TP, it publishes on the /joy topic. Closing this issue would mean, that we are able to drive/steer the magni robot with it.

mjstn, 2020-09-22:
I also don't understand what this issue is about.
In the last 3 months and one from last month I have bought 2  LogiTech wireless F710 joysticks.
You put the dongle in the Pi and all the 'magic' to get speed on /cmd_vel all works.
You can take or default image or even a new Avalon one, no settings are required to be changed.

What exactly is the issue here?

tp4348, 2020-09-22:
I addressed it wrong. I was testing it on Magni in TP that probably no one touched for the last 5-6 months. The thing is, the Magni wasn't moving, although the joystick published on /joy topic. I might have missed something because it was my first time doing with the actual physical robot(However, I got it to drive with keyboard). I will close this issue, according to the fact the problem is more specific to the robot in TP (or to me, I guess).

closed 2020-09-22 by tp4348