UbiquityRobotics/magni_robot #140: [magni_description] Raspicam description does not follow ROS Coordinate Conventions

eupedrosa, 2021-01-04:
## Summary

The `raspicam` sensor description (i.e. frames) does not not follow ROS Coordinate Conventions [REP 103](https://www.ros.org/reps/rep-0103.html).
From the REP:
> ### Suffix Frames
>In the case of cameras, there is often a second frame defined with a "_optical" suffix. This uses a slightly different convention:
>
>z forward
>x right
>y down

Not following this convention can introduce unwanted results when using metric information extracted from the image. For example, the [fiducial detection package](https://github.com/UbiquityRobotics/fiducials/tree/kinetic-devel/aruco_detect) return a the pose of the marker on the aforementioned coordinate system. The transformation of this pose to any other frame would result in a wrong pose.

## Example
### Using the current description:
[image]

The marker pose, w.r.t. the robot's frame, is *incorrect*!

### Using a description that follows the ROS Coordinate Conventions:
[image]
The marker pose, w.r.t. the robot's frame, is now *correct*!

## How to fix ?

The quick fix would be to change https://github.com/UbiquityRobotics/magni_robot/blob/cfdfeed95ad88077adf3a15501f3a34389593dfa/magni_description/urdf/sensors/raspi_camera.xacro#L34
to
```xml
<origin xyz="0 0 0.012" rpy="${-pi/2} 0 ${-pi/2}" />
```

However, please note that cameras description usually have 3 tf frames: prefix > prefix_rgb_frame > prefix_rgb_optical_frame

Polonium19, 2021-01-29:
@eupedrosa  Thank you for contaciting with the issue.
The link `raspicam_sensor` is used only for Gazebo simulation to align Gazebo camera with link `raspicam`.
This additional link is required because Gazebo camera sensor is oriented following X-axis.
The real camera link `raspicam` is oriented following Z-axis (as it sould be according to the ROS convention). 
Use only `raspicam` link for camera positioning.
I will do some changes to the URDF to eliminate this confusion.

JanezCim, 2021-02-02:
@eupedrosa  i've tried out the @Polonium19 s implementation and now it looks something like this:
[image]
(front one named "raspicam_gazebo_sensor" and back one named "raspicam")

If we wont have "_sensor" and "_optical" frames for real camera then i think this is ok solution. In this case we just set the raspicam driver to post with "frame_id=raspicam". What do you think?

eupedrosa, 2021-02-09:
In my understanding of the ROS convention, there should always be a "_sensor" or an  "_optical" frame. That is why Gazebo requires the link `raspicam_sensor`. Gazebo is emulating the actual physical sensor that has its own coordinate system.

We usually use the word "sensor" to specify two different things:
1. The sensor box, for example, a camera that holds electronics.
2. The optical sensor that captures the light going throught a lens and creates a bitmap.

Hence, we allways have at least two frames, one for the camera and another for the actual optical sensor.

@JanezCim, Was I clear with my explanation? We may talk if you still have some doubts.

closed 2021-04-06 by JanezCim