UbiquityRobotics/magni_robot #142: magni_description description.launch

anfederman, 2021-01-15:
<node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />

The last Line needs to be changed to:

 <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />

In order to work in Melodic and Noetic - generates a warning or fails to render model correctly in rviz/gazebo.

For some reason, I can't created mu own pull request.

MoffKalast, 2021-01-15:
Oh wow, this may be a thing in kinetic as well. I've had that happen on one of my installations but figured I set something up wrong. I'll run a short test and make the change if it works fine on kinetic as well.

MoffKalast, 2021-01-15:
Hmm, well it doesn't fix that specific issue but at least I can confirm that it makes no difference in kinetic. I'll make a PR.

MoffKalast, 2021-01-15:
Actually, let me set this as closed by the PR itself when it's merged.

PR #144 (2021-01-18): Noetic/Melodic description compatibility change for state_publisher.
Closes #142

As far as I can tell the change makes no difference on kinetic, so there shouldn't be any backwards compatibility problems. @JanezCim I think you have a melodic installation somewhere so I guess you can check if this makes any difference at some point.
closed 2021-01-18 by MoffKalast