UbiquityRobotics/magni_robot #148: Pi4 Ethernet cannot get Ipv4 assignment if plugged into MCB

mjstn, 2021-02-12:
Have been running into a problem very consistently on one particular Pi4 but I don't have good sample size on other Pi4.
Issue is I get a new image (in particular 1-25-2021) and the Pi can fully run a magni and can work on WiFi but will not come up and get an IP from my network  EVER.   

Insight into no IP addr on Pi4  on the 1-25-2021 image on just ONE of my Pi4 boards it seems.    

I found that dmesg does show on the Pi eth0 going down when I unplug cable and up when I plug it in.  Likewise ifconfig only shows inet6 addr when cable plugged in.  This is ipv4 and/or DHCP related perhaps and is brought on by connection to the MCB.

Next I found I can in fact ping from windows on my same home router using 'ping -l en1  <inet6_addr>

My router thinks the mac for this Pi4 is off.  That is likely because I had gotten the IP v4 using a different image that does allocate an IP an hour or so before this.

This makes me think there is something not right with the Pi4 negotiating with the DHCP server of my 3 year old ATT U-Verse router.  

Another HUGE clue is that I stop the Pi4, unplug it from the MCB.
Then with very same lan cable and so on plug the Pi into a 2.4Amp USB supply and Ip gets assigned.

This is very disturbing.  WiFi works when plugged into MCB, just Ethernet is wonky

mjstn, 2021-02-12:
Another HUGE clue.   If I make a good solid adapter to connect just the 2 +5 lines and two of the grounds from the MCB to the Pi4 it gets onto the network just fine.

I will now investigate if this is due to proximity of the Enet interface to the board OR is due to other lines connected to the MCB.

MoffKalast, 2021-02-12:
Just a thought, is this only DHCP related or is it also a problem with a static IP? If so it may be part of the problem we've having with the lidar communications on occasion.

mjstn, 2021-02-12:
No clue.   Just reporting this from standpoint of a very serious ethernt issue.  Wifi works fine.   Add to notes as observations are found.

mjstn, 2021-02-12:
Major clues:
- It also works when adding 3.3V but that is a no connect on MCB so so surprise there
- It does NOT work if I add in  I2C bus
- MAJOR clue!    A fully plugged in Pi4 works on ethernet if I remove the RTC from the MCB

I will note that I had been using the 3V battery so this is not as simple as the old 'no battery messes up things' issue.

I am capturing dmesg outputs from working and failed cases as the next step here

mjstn, 2021-02-13:
From dmesg output we see these entries relating to RTC init when we have failure the MCP7940 is installed.

With MCP7940:    2.535865] rtc-ds1307 1-006f: registered as rtc0
No   MCP7940:     2.369936] rtc-ds1307: probe of 1-006f failed with error -121

Using a Pi3 we also see the 'rtc-ds1307 1-006f: registered as rtc0' message and that works fine.

We do not know if the presense of the RTC is responsible itself OR if that in some way impacts timing to lead to the fault.

mjstn, 2021-02-13:
Well I really hate to resort to such tactics but I do have a fix although I cannot 'justify it with paperwork'.
**Modify /boot/config.txt  and change   boot_delay=2**

The default is 1 by the way.    This has worked 4 times so far and it NEVER worked before.

MoffKalast, 2021-02-13:
Seems legit.

mjstn, 2021-02-13:
Vid:   Can you please do this on Avalon images too?

JanezCim, 2021-02-15:
The boot_delay seems to be helping with normal magni robots, restarted about 7 times now without problems - wheres before it failed to boot in about 1/5

rohbotics, 2021-02-15:
New testing image with the boot delay built in: https://ubiquity-pi-image.sfo2.cdn.digitaloceanspaces.com/2021-02-15-ubiquity-xenial-lxde-rpi-testing.img.xz

MoffKalast, 2021-02-15:
> wheres before it failed to boot in about 1/5

I'm sorry, it WHAT. Wait you're saying this fixes the baud rate bug as well?

JanezCim, 2021-02-15:
in about 1 out of 5 times I could not move motors on reboot and i saw this output.

[image]

MoffKalast, 2021-02-16:
@mjstn well it appears this fix also solves the baud rate bug somehow.

rohbotics, 2021-02-16:
Well that's convenient :grinning:

mjstn, 2021-02-17:
In general I am not real excited about claiming things are 'fixed' by a delay without an explanation of why or what the delay does to lessen chance of the root cause rearing its ugly head. 

I am excited that we can greatly at least ease the trouble by use of such 'ugly'  blind delay techniques.

So I guess I'm just saying this to justify my discovery of this hack-workaround.

JanezCim, 2021-05-07:
@mjstn i added this hack to the todo change list for the new image that we will release publicly in #163 and referenced this issue there as well, so we might as well close this if you are not planning to do a deeper dive into solving it.

JanezCim, 2022-04-20:
Closing this because of inactivity and basically it was solved

closed 2022-04-20 by JanezCim