UbiquityRobotics/magni_robot #149: Fleet manager to integrate with RMF

JGUO006, 2021-02-17:
Hi,

May I know whether there is any package called fleet manager provided to integrate with RMF or RoMi-H? I have tried the office demo world under osrf/rmf_demo repository and wonder whether I can run the actual Magni robot and complete a simple delivery request instead of the simulated robot in office demo environment. May I know how I can integrate RMF (rmf_core, fleet manager, fleet adapter, etc) so that the actual robot can do the same as shown in the demo?

JanezCim, 2021-09-22:
Hello @JGUO006 sorry for the large delay. I dont know if anyone here internally already integrated Magni code into the open_rmf framework but maybe somebody in the community has, could you also ask this question on UR discord channel: https://forum.ubiquityrobotics.com/ 
But it would be interesting idea to have Magni integrate into that! Will suggest this to people

JunHaoChng, 2022-03-16:
Hello! If you are still interested in integrating the magni to RMF (Romi-h is the same thing as RMF), you can check out this repo here: https://github.com/sharp-rmf/magni/tree/missybot

Everything is a little bit rough, because we used them for ours demos but didn't bother to clean them up afterwards. If you do have questions do let me know I will be happy to answer.

The main way of integrating Magni to RMF is via using the free_fleet (reference below), the free fleet client is ran on the magni, and the free fleet server is ran on the computer running rmf_core. 

Free_fleet
https://github.com/open-rmf/free_fleet
How to use free fleet with RMF thread here:
https://github.com/open-rmf/free_fleet/issues/75

You can check out the launch file I used for magni with the free fleet server side here:
https://github.com/sharp-rmf/chart_demos/blob/main/sutd/sutd_launch/launch/sutd_9_Feb_2022.launch.xml#L5-L32

Free fleet client side, which is actually ran on the magni itself:
https://github.com/sharp-rmf/magni/blob/missybot/magni/launch/magni_45_free_fleet_client.launch

JanezCim, 2022-03-16:
Hello, thank you very much for the suggestion, we still haven't gotten around to trying RMF out although we are keeping it on the TODO list. Will also look at your suggestions when we come around to it.