UbiquityRobotics/magni_robot #150: May need to account for git credential authentication on future images

mjstn, 2021-02-21:
Need to understand and try the git authentication based repository access on our images.
It is likely to be fine but as of Aug 13 2021 (A Friday by the way) user and password access to our private repositories will have to be using authentication tokens.

https://github.blog/2020-12-15-token-authentication-requirements-for-git-operations/

rohbotics, 2021-02-21:
This shouldn't be an issue for the public images, as there are no authenticated pulls there. Not sure if any of the customer stuff still pulls directly from github, or if it all goes through the update system @MoffKalast

MoffKalast, 2021-02-22:
Well it's gonna be more of an issue on our development side, but otherwise not really I think. Customers either have their own accounts or update system access.