UbiquityRobotics/magni_robot #152: mdmsum when connected via ssh

anfederman, 2021-03-06:
[ERROR] [1615052244.401622947]: Client [/rostopic_7441_1615052235674] wants topic /battery_state to have datatype/md5sum [sensor_msgs/BatteryState/476f837fa6771f6e16e3bf4ef96f8770], but our version has [sensor_msgs/BatteryState/4ddae7f048e32fda22cac764685e3974]. Dropping connection.

This happens when the robot is running on Noetic, but I am running my laptop at 18.04 melodic.   I can drive the robot using keyboard teleop.   Also, rc.local names aren't working.

I am setting ROS_IP and ROS_MASTER_URI to actual IP numbers instead of "name.local"

If I don't get this error message. I just get no data back on the rostopic echo request.

I suspect this is either due to rc.local confusion, or problems with the lack of a working pifi on booting via magni-base and launch_core.py

anfederman, 2022-07-20:
closed as inactive

closed 2022-07-20 by anfederman

MoffKalast, 2022-07-28:
Yeah no that's due to it being a different ROS version iirc, I always get these hash mismatches when accidentally opening Kinetic bags under Noetic or vice versa. As a rule of thumb, Kinetic and Noetic are compatible because the hashes didn't change for most messages, while Noetic isn't because of the larger amount of these.