UbiquityRobotics/magni_robot #153: License Text Missing

barulicm-ms, 2021-03-11:
While all of the packages in this repository list their license as "BSD" in their package manifest, the actual text of the license seems to be missing from this repository.

Please consider adding a LICENSE file including the full text of the license under which this code is released. This will allow other organizations to modify and distribute (or not) this code as you intend.

Thank you!

JanezCim, 2021-03-25:
is meantioning BSD inside 
https://github.com/UbiquityRobotics/magni_robot/blob/1a4a8a2a30b6ff0c089c40ddf0d393385a78381d/magni_bringup/package.xml#L8
not enough? Why would we need to include the full text? Is it suggested by any REP (looking at [REP140](https://www.ros.org/reps/rep-0140.html#description) this is not mentioned anywhere)

do you have and idea @rohbotics or @MoffKalast ?

barulicm-ms, 2021-03-25:
We are asking that same question about the package manifest to our legal team.

I'm certainly not a lawyer or legal expert by any means. The strongest point of evidence I could point to that would suggest the repository should include the full text of the license is the text of the BSD license itself. The license claims that any redistribution (source or binary) must retain or reproduce the license text. 

I am referencing the 3-clause BSD license text available on opensource.org (https://opensource.org/licenses/BSD-3-Clause).

closed 2021-04-30 by JanezCim