UbiquityRobotics/magni_robot #155: Magni needs OLED node to come up by default on image and be included in apt upgrade

mjstn, 2021-03-15:
As title says we need these two things for the Magni product line (includes breadcrumb but Magni robot is #1 priority)

1. get to Jenny in production proper images that have current OLED display actually set to come up.  The node ignores display if not found.
2. have things so an 'apt upgrade' pulls in OLED display node as well as whatever launch changes are required so   'magni-base' service includes firing it up.

anfederman, 2021-03-25:
This is already implemented in my SD image.   Changes were made to the /etc/ubiquity/robot.yaml   and /magni_robot/magni_bringup/ base launch.py   -  also  oled_display_node files.

anfederman, 2021-03-25:
The only change needed (I think)  is to the robot.yaml file

ubuntu@magnitest:/etc/ubiquity$ more robot.yaml :+1: 
--------
# Robot Configuration

raspicam: {'position' : 'front'}

#sonars: None
sonars: 'pi_sonar_v1' # Use this to enable sonars

#OLED display

oled_display: {'controller' : 'SH1106'}

------

MoffKalast, 2021-03-26:
Ok that's good to know, we should add that to the script that generates the yaml.

closed 2021-07-26 by MoffKalast