UbiquityRobotics/magni_robot #163: Todo list for new ubiquity-xenial-lxde RPI image release

JanezCim, 2021-05-07:
We are making updates on magni_robot repo, but some changes are going to be needed to be included in the new public image release (on https://downloads.ubiquityrobotics.com/pi.html). This is a placeholder for those changes that need to happen. Please feel free to add stuff to this list

- [ ] Include `source <path_to_scrpit>/ros_log_clean.sh` into /usr/sbin/magni-base and comment it out by default. Make a readme (maybe in home directory) that explains that if you need log cleaning you should uncomment that. #161 
- [ ] update, upgrade?
- [ ] check if robot.yaml is up to date

JanezCim, 2021-05-07:
@MoffKalast do you have something to add to this list?

MoffKalast, 2021-05-07:
You should probably make this as a PR in https://github.com/UbiquityRobotics/pi_image_build, after all that's the config that has the buildbot generate the actual stock image. Not really magni_robot related.

JanezCim, 2021-05-07:
ok, didnt know that exsisted, will do a PR there, thanks!

JanezCim, 2021-05-07:
Boot delay has already been fixed here: 
https://github.com/UbiquityRobotics/pi_image_build/blob/b5a0343018876410b1236d195324051bb37027cf/files/config.txt#L660

and ive added the logclean file option into a PR in that repo, so closing this

closed 2021-05-07 by JanezCim