UbiquityRobotics/magni_robot #169: Automated test for launch_core

rohbotics, 2021-07-22:
We should write some automated tests to make sure that launch_core is functional, maybe just running it with `--debug` and various settings to make sure that the outputted launch file is reasonable.

@JanezCim Thoughts?

JanezCim, 2021-07-23:
That would def be a good idea. Even a necessary one since its super important to get it right if changes happen. I can start working on this.

JanezCim, 2021-07-26:
ok so quoting from slack

> Launch files in ros2 are written in python, so launch_core will be much much nicer as we will no longer have to even generate the XML, just deal with it semantically.

ok so since noetic seems to be our last ROS1 release, and ROS2 wont need a XML generation, would be smart to dedicate a huge effort to automating this even? Maybe if we plan to support Noetic for the next 5 years then a smaller scale test would be better?

rohbotics, 2021-07-26:
I think checking that it generates valid XML is a good start, and we should definitely do that as a minimum. Maybe we could do that with a few different configuration options as well. Anything more than that is probably too much effort for not much reward (we will still be manually testing anyway) considering the upcoming ROS2 migration.

JanezCim, 2021-07-27:
Yeah okay but how would you check of the XML is valid? try to roslaunch it and if it fails its not ok? Is there any other ROS-way to do the validation?

rohbotics, 2021-07-27:
The best option is probably running `roslaunch check` https://wiki.ros.org/roslaunch/Commandline%20Tools#roslaunch-check

JanezCim, 2021-09-10:
Yeah, okay, so some update on this: the problem is that roslaunch-check works only if `abc.launch` is inside the `package/launch/abc.launch` because when checking for launch dependancies it looks at where the actual location of launch file is:

When I try to launch it from /tmp/:
```
ubuntu@pi-focal ~/catkin_ws/src/magni_robot/magni_bringup (bugfix-oldrobotyaml) $ rosrun roslaunch roslaunch-check /tmp/core.launch
Traceback (most recent call last):
  File "/opt/ros/noetic/share/roslaunch/scripts/roslaunch-check", line 93, in <module>
    pkg_dir = r.get_path(pkg)
  File "/usr/lib/python3/dist-packages/rospkg/rospack.py", line 207, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
rospkg.common.ResourceNotFound: <exception str() failed>
```
But when I try moving to magni_bringup/launch it works: 
```
ubuntu@pi-focal ~/catkin_ws/src/magni_robot/magni_bringup (bugfix-oldrobotyaml) $ rosrun roslaunch roslaunch-check launch/core.launch 
checking launch/core.launch
xacro: in-order processing became default in ROS Melodic. You can drop the option.
...writing test results to /home/ubuntu/.ros/test_results/magni_bringup/rosunit-roslaunch_check_launch_core_launch.xml
passed
```
This means we would need to move the core.launch generation to package magni_bringup/launch - but that would not be ok because it would always be spamming on git change log. Other option is that we then put the core.launch into gitignore. What do you think? Any ideas?

PR #177 (2021-12-24): Slightly improved architecture of robot.yaml
Changes in how core.launch gets generated and launched. Biggest change: base.yaml does not exsist any more, everything gets loaded through magni_bringup/config/default_robot.yaml -> launch_core.py -> generated_core.launch. This way the params from base.yaml are not duplicated and hopefully makes the whole process more clear.

Fixes #172 
Fixes #169 

**on the rpi that runs noetic** test with 
`python3 ~/catkin_ws/src/magni_robot/magni_bringup/scripts/launch_core.py --debug`

when reviewing: 
 - go over the code to see if there are any stupid mistake [... cut]
closed 2021-12-24 by JanezCim