UbiquityRobotics/magni_robot #170: Firmware not reporting version, rejecting communication

cst0, 2021-07-26:
I'm trying to run this package from source after finding that the default image isn't quite what we're looking for, and I've run into a bit of trouble here. Running the command `roslaunch magni_bringup base.launch` fails to actually work: a series of warnings reading 

```
...
[ WARN] [1627669152.359458507]: REJECT 20
[ WARN] [1627669152.359646745]: REJECT fc
[ WARN] [1627669152.360869040]: REJECT 20
[ WARN] [1627669152.361051829]: REJECT fc
[ WARN] [1627669152.409428438]: REJECT 20
[ WARN] [1627669152.409664936]: REJECT fc
...
```

repeat, with the occasional

```
[ERROR] [1627669152.257516563]: The Firmware not reporting its version
```

I assume this is related to an error that presents itself before the ROS content launches, reading:
```
Error reading motor controller board version from i2c
```

But I have confirmed that the i2c bus is available, there are things on the i2c bus, and the serial port being requested is also available. I've manually enabled both of these and have verified that they're up via `i2cdetect` and `screen` respectively (although I'm not seeing data come over via screen, though that may not mean anything since I don't know what I should be seeing here).

Any further documentation or help would be much appreciated!

cst0, 2021-07-26:
Some more investigating: the error messages are being produced by the motor_node within ubiquity_motor, which I had actually installed via the binaries for ros-melodic. I was unable to get the test_motor_board to run on that version due to a typo in the execution path of that file, so I tried instead installing that package from source as well: I'm seeing the same error here in the motor_node.

This does not appear to be related to https://github.com/UbiquityRobotics/ubiquity_motor/issues/139, because checking out that branch does not resolve these issues.

cst0, 2021-07-26:
Found this [thread](https://forum.ubiquityrobotics.com/t/motor-node-fails-with-firmware-not-reporting-its-version/183/5), but the associated config files discussed there didn't work after a reboot. I've also now tried both the most recent commits on the kinetic-devel and the noetic-devel branches on both this repo and the ubiquity_motor repo, so I'm now running through the [Troubleshooting](https://learn.ubiquityrobotics.com/troubleshooting) page that was linked in the thread.

cst0, 2021-07-26:
Unfortunately I haven't had any luck with that. This seems to be the "rare baud rate bug", but rebooting certainly does not get rid of it. It seems odd to me that the page marks it as resolved, as the link to resolve the error through updating does not actually link to a page with any content about updating (and I'm updated to the extent that that's possible anyway).

mjstn, 2021-07-26:
A bit late getting to this.   Hello Chris.   Please tell me what the version of your MCB (main board) is.   This is key to possible solutions.   Boards from the last year have MCB board rev in nice big white silkscreen on left edge of the board as you face the robot.     Older boards can be VERY hard to read the board rev (in same place) so I have this page:   https://learn.ubiquityrobotics.com/PC_Board_RevId

If you are doing your own thing and NOT using an MCB this is CRITICAL to tell me now as that is an entirely different 'ball game' of possible serial comm issues. 

Next if you do have the two leds at top middle labeled   'SIN' and 'SOUT' please tell me if both ever start blinking?   The SOUT led will blink if the on-board processor is outputting its periodic status reports.  If  SOUT does not blink there is a hardware MCB issue or some firmware issue on the MCB. 

Lastly tell me what Ubiquity Image you started with OR if it is same image that came from the robot and if so when did you get this robot?

Lastly if you do have the 'rare baud rate bug'  the 'signature' for that is tons of corruption errors showing up in our log.   Our log is located here:     /home/ubiquity/.ros/log/latest/rosout.log      Also if you have and know how to use an oscilloscope you can see the 'rare baudrate bug'   by looking with a scope the serial out of the MCB on the FTDI jack, P508, on the TX pin, pin 4.  The proper 38.4k baud would show the shortest pulses you see (1 bit time) to be around 26 microSeconds.    This is an FYI, I don't expect you to try this unless you are comfortable with this sort of method.

Thanks,
Mark

cst0, 2021-07-28:
Thanks-- this is running on the 5.3, although I'm fairly confident it was a software issue on my end. I'll close this since I've resolved it, but I'll add more detail just in case anyone else comes across this thread.

Because I've been adding some additional sensing hardware and other resource-intensive software to the Magni for our use-case, we decided it made sense to try to offload some of that computation to a device networked via ethernet. So this meant needing to connect the provided Pi, the ethernet LIDAR unit, and the new device via ethernet through a passive networking switch. I was having trouble implementing this and our other networking requirements on the existing image, so I decided to spin up an Ubuntu ARM image and configure it for ROS with the magni packages. The only change was the image, which is what makes me confident it was on my end (the original image still worked).

I believe the error came from my not testing in a fully clean environment: by not shutting down the background magni service consistently between testing configurations, the serial port and serial baud ROS parameters remained in the parameter server and led to a mismatch between intended software version and intended hardware version.

The fix has been to better analyze the default image's networking configuration to adapt it rather than starting from scratch. The ethernet port of the Pi seems to need to be assigned the address 192.168.42.125 in order to communicate with the LIDAR, so I've configured Dnsmasq on the Pi to assign addresses on the 192.168.42.1/24 subnet, and assigned all devices on the network appropriate static IP's in the same way. I had been incorrectly assuming that the ethernet interface IP addresses could be assigned more arbitrarily, but being stricter to that subnet has allowed for typical ethernet communication.

closed 2021-07-28 by cst0