UbiquityRobotics/magni_robot #174: magni_gazebo launch files not working

JanLipovsek, 2021-08-20:
Magni_gazebo launch files cant load robot_description parameter 

[roslaunch][INFO] 2021-08-16 17:07:11,371: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2021-08-16 17:07:11,376: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2021-08-16 17:07:11,376: roslaunch starting with args ['/opt/ros/noetic/bin/roslaunch', 'magni_gazebo', 'cafe2_world.launch']
[roslaunch][INFO] 2021-08-16 17:07:11,376: roslaunch env is environ({'SHELL': '/bin/bash', 'ROS_VERSION': '1', 'SESSION_MANAGER': 'local/lipa-PC:@/tmp/.ICE-unix/2214,unix/lipa-PC:/tmp/.ICE-unix/2214', 'QT_ACCESSIBILITY': '1', 'COLORTERM': 'truecolor', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'XDG_MENU_PREFIX': 'gnome-', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'PKG_CONFIG_PATH': '/home/lipa/tutorials/devel/lib/pkgconfig:/home/lipa/ubiquity/usb_cam/devel/lib/pkgconfig:/home/lipa/ubiquity/track_objects/devel/lib/pkgconfig:/home/lipa/ubiquity/dnn_detect/devel/lib/pkgconfig:/home/lipa/ubiquity/magni/devel/lib/pkgconfig:/home/lipa/ros1/devel/lib/pkgconfig:/home/lipa/catkin_ws/devel/lib/pkgconfig:/opt/ros/noetic/lib/pkgconfig', 'ROS_PYTHON_VERSION': '3', 'LC_ADDRESS': 'sl_SI.UTF-8', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'LC_NAME': 'sl_SI.UTF-8', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'XMODIFIERS': '@im=ibus', 'DESKTOP_SESSION': 'ubuntu', 'LC_MONETARY': 'sl_SI.UTF-8', 'SSH_AGENT_PID': '2179', 'ROS_PACKAGE_PATH': '/home/lipa/tutorials/src:/home/lipa/ubiquity/usb_cam/src:/home/lipa/ubiquity/track_objects/src:/home/lipa/ubiquity/dnn_detect/src:/home/lipa/ubiquity/partybot/src:/home/lipa/ubiquity/magni/src:/home/lipa/ros1/src:/home/lipa/catkin_ws/src:/opt/ros/noetic/share', 'GTK_MODULES': 'gail:atk-bridge', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/lipa/tutorials/devel/share/common-lisp:/home/lipa/ubiquity/usb_cam/devel/share/common-lisp:/home/lipa/ubiquity/track_objects/devel/share/common-lisp:/home/lipa/ubiquity/dnn_detect/devel/share/common-lisp:/home/lipa/ubiquity/partybot/devel/share/common-lisp:/home/lipa/ubiquity/magni/devel/share/common-lisp:/home/lipa/ros1/devel/share/common-lisp:/home/lipa/catkin_ws/devel/share/common-lisp', 'DBUS_STARTER_BUS_TYPE': 'session', 'PWD': '/home/lipa/catkin_ws/src/magni_robot/magni_gazebo/launch', 'LOGNAME': 'lipa', 'XDG_SESSION_DESKTOP': 'ubuntu', 'XDG_SESSION_TYPE': 'x11', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'WINDOWPATH': '2', 'HOME': '/home/lipa', 'USERNAME': 'lipa', 'IM_CONFIG_PHASE': '1', 'LC_PAPER': 'sl_SI.UTF-8', 'LANG': 'en_US.UTF-8', 'ROS_ETC_DIR': '/opt/ros/noetic/etc/ros', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'VTE_VERSION': '6003', 'CMAKE_PREFIX_PATH': '/home/lipa/tutorials/devel:/home/lipa/ubiquity/usb_cam/devel:/home/lipa/ubiquity/track_objects/devel:/home/lipa/ubiquity/dnn_detect/devel:/home/lipa/ubiquity/partybot/devel:/home/lipa/ubiquity/magni/devel:/home/lipa/ros1/devel:/home/lipa/catkin_ws/devel:/opt/ros/noetic', 'GNOME_TERMINAL_SCREEN': '/org/gnome/Terminal/screen/f5476700_e9f7_460c_a81e_926b7fc227d9', 'INVOCATION_ID': '8d28051196f842f7b926a1e175a8f9e8', 'MANAGERPID': '1999', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'XDG_SESSION_CLASS': 'user', 'PYTHONPATH': '/home/lipa/tutorials/devel/lib/python3/dist-packages:/home/lipa/ubiquity/track_objects/devel/lib/python3/dist-packages:/home/lipa/ubiquity/dnn_detect/devel/lib/python3/dist-packages:/home/lipa/ros1/devel/lib/python3/dist-packages:/home/lipa/catkin_ws/devel/lib/python3/dist-packages:/opt/ros/noetic/lib/python3/dist-packages', 'TERM': 'xterm-256color', 'LC_IDENTIFICATION': 'sl_SI.UTF-8', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'USER': 'lipa', 'GNOME_TERMINAL_SERVICE': ':1.107', 'DISPLAY': ':1', 'SHLVL': '1', 'LC_TELEPHONE': 'sl_SI.UTF-8', 'ROS_MASTER_URI': 'http://localhost:11311', 'QT_IM_MODULE': 'ibus', 'LC_MEASUREMENT': 'sl_SI.UTF-8', 'DBUS_STARTER_ADDRESS': 'unix:path=/run/user/1000/bus,guid=5badc050eb2f39b02e782f02611a4ca3', 'ROS_HOSTNAME': 'lipa-PC.local', 'LD_LIBRARY_PATH': '/home/lipa/tutorials/devel/lib:/home/lipa/ubiquity/usb_cam/devel/lib:/home/lipa/ubiquity/track_objects/devel/lib:/home/lipa/ubiquity/dnn_detect/devel/lib:/home/lipa/ubiquity/partybot/devel/lib:/home/lipa/ubiquity/magni/devel/lib:/home/lipa/ros1/devel/lib:/home/lipa/catkin_ws/devel/lib:/opt/ros/noetic/lib', 'XDG_RUNTIME_DIR': '/run/user/1000', 'LC_TIME': 'sl_SI.UTF-8', 'JOURNAL_STREAM': '8:50437', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'PATH': '/opt/ros/noetic/bin:/home/lipa/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/lipa/home/lipa/clion/clion-2020.1.1/bin:/home/lipa/pycharm-community-2020.3.5/bin', 'GDMSESSION': 'ubuntu', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus,guid=5badc050eb2f39b02e782f02611a4ca3', 'ROS_ROOT': '/opt/ros/noetic/share/ros', 'ROS_DISTRO': 'noetic', 'LC_NUMERIC': 'sl_SI.UTF-8', 'OLDPWD': '/home/lipa/catkin_ws', '_': '/opt/ros/noetic/bin/roslaunch', 'ROS_LOG_FILENAME': '/home/lipa/.ros/log/f5119c7a-fea0-11eb-8f41-5918bd42ecc4/roslaunch-lipa-PC-30657.log'})
[roslaunch][INFO] 2021-08-16 17:07:11,376: starting in server mode
[roslaunch.parent][INFO] 2021-08-16 17:07:11,376: starting roslaunch parent run
[roslaunch][INFO] 2021-08-16 17:07:11,377: loading roscore config file /opt/ros/noetic/etc/ros/roscore.xml
[roslaunch][INFO] 2021-08-16 17:07:11,495: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2021-08-16 17:07:11,496: loading config file /home/lipa/catkin_ws/src/magni_robot/magni_gazebo/launch/cafe2_world.launch
[roslaunch][INFO] 2021-08-16 17:07:11,537: Added node of type [gazebo_ros/gzserver] in namespace [/]
[roslaunch][INFO] 2021-08-16 17:07:11,537: Added node of type [gazebo_ros/gzclient] in namespace [/]
[roslaunch][ERROR] 2021-08-16 17:07:11,794: RLException: while processing /home/lipa/catkin_ws/src/magni_robot/magni_gazebo/launch/magni.launch:
while processing /home/lipa/catkin_ws/src/magni_robot/magni_description/launch/description.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '--inorder', '/home/lipa/catkin_ws/src/magni_robot/magni_description/urdf/magni.urdf.xacro', 'sonars_installed:=true', 'lidar_extrinsics_file:=/home/lipa/catkin_ws/src/magni_robot/magni_description/param/lidar_extrinsics_top_plate.yaml', 'camera_extrinsics_file:=/home/lipa/catkin_ws/src/magni_robot/magni_description/param/camera_extrinsics_forward.yaml']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro --inorder         '$(find magni_description)/urdf/magni.urdf.xacro'          sonars_installed:=$(arg sonars_installed)         lidar_extrinsics_file:=$(arg lidar_extrinsics_file)         camera_extrinsics_file:=$(arg camera_extrinsics_file)         "/>
[... cut ...] _ws/src/magni_robot/magni_description/param/camera_extrinsics_forward.yaml']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro --inorder         '$(find magni_description)/urdf/magni.urdf.xacro'          sonars_installed:=$(arg sonars_installed)         lidar_extrinsics_file:=$(arg lidar_extrinsics_file)         camera_extrinsics_file:=$(arg camera_extrinsics_file)         "/>

[rospy.core][INFO] 2021-08-16 17:07:11,798: signal_shutdown [atexit]

anfederman, 2021-08-20:
> Invalid tag: Cannot load command parameter [robot_description]: command
> roslaunch.xmlloader.XmlParseException: Invalid tag: Cannot load command

Noetic/Ubuntu 20.04 doesn't support Gazebo on a Raspberry PI.  There was no
deb for Gazebo on a pi Unbutu 20.04. I am not sure, but I don't think it
exists for either ARM32 or ARM64.  At any rate, there isn't a desktop-full
for the ARM32 Pi ubuntu.

Magni-gazebo worked fine on Pi running 18.04 melodic.  I will try to
install magni_gazebo on an AMD64 Ubuntu and see if it works.

anfederman, 2021-08-20:
I have tested magni_gazebo empty_world launch on my Ubuntu 20.04 AMD64
laptop.  It works well.
I don't think we need to worry about getting this working on a Raspberry PI.
[image: image.png]

JanLipovsek, 2021-08-20:
Yeah i forgot to mention in my first post that I was trying to run the noetic_devel branch of magni robot on my PC runing ubuntu 20.04. The indigo_devel branch works OK.

JanezCim, 2021-09-01:
@JanLipovsek  noetic_devel branch is still very much in development, it is in now way a release. There has been a huge reconstruction of how gazebo launches and bringup files are ran at the boot of the robot. So that branch is very prone to errors which we are trying to solve. You can read how the new bringup works here: https://github.com/UbiquityRobotics/magni_robot/blob/noetic-devel/magni_bringup/README.md

anfederman, 2021-09-01:
I am guessing since magni_gazebo won't work on a Pi under noetic, it won't be released on the final image?   It needs to be broken out as a separate installation. If we intend to continue to support gazebo.

Also if we intend to support it for workstations, we need to modify it to support fiducials.  empty_world launch works ok on my laptop, but the fiducial_worlds doesn't show the ceiling markers either on rviz or gazebo.

JanezCim, 2021-09-07:
@anfederman afaik till now magni_gazebo was always included within the Pi image even though it was never used there. A bit of dead weight, for downloading and compiling. We could put it in a separate repo but then you have another thing to support and you have to probably change some READMEs - what is your opinion here @rohbotics ?

Im taking a look at the fiducial ceiling markers now.

JanezCim, 2021-09-07:
Fiducial ceiling markers should be solved with https://github.com/UbiquityRobotics/magni_robot/pull/176

anfederman, 2021-09-07:
Sorry  after applying 176:

xacro: in-order processing became default in ROS Melodic. You can drop the
option.
[Errno 2] No such file or directory:
'/home/dr1bot/catkin_ws/src/magni_robot/magni_description/param/camera_extrinsics_forward.yaml'

  I needed to copy the directory 'extinsics' to 'param'  and then empty
world worked ok on my laptop noetic.

I think you nan fix by editing the magni.launch  or by copying the
directory, there is a inconsistency that insome cases the extrinsics are
under param and in other places under extrinsics.

anfederman, 2021-09-07:
Also , fiducial world launches, if you edit the launch and change the
camera to:

<arg name="camera_position" default="forward"/>   (as opposed to
raspicam_mount)

However the map isn't launched and the ceiling fiducials aren't being
displayed.

JanezCim, 2021-09-09:
> I think you nan fix by editing the magni.launch  or by copying the directory, there is a inconsistency that insome cases the extrinsics are under param and in other places under extrinsics.

This was corrected here: https://github.com/UbiquityRobotics/magni_robot/blob/f5560cdef82d67f2936ee2677cd88be1105cdc49/magni_gazebo/launch/magni.launch#L22

> <arg name="camera_position" default="forward"/>   (as opposed to raspicam_mount)

This has been corrected here: https://github.com/UbiquityRobotics/magni_robot/blob/f5560cdef82d67f2936ee2677cd88be1105cdc49/magni_gazebo/launch/fiducials/fiducial_world.launch#L4

I've had exactly the same problems that you are having and solved them in that branch along with the fiducials not being displayed. So im not sure where something is going wrong but i suspect your git is not pulling correctly or you are launching a different launch file. Can you copy-paste the command that you are launching this with? You are launching on a PC and NOT on a RPI right?

anfederman, 2021-09-09:
roslaunch magni_gazebo fiducial_world.launch

everything works, no errors except no upward markers seen on gazebo.

further these warnings:

, set to ""
[ WARN] [1631214085.736891296, 0.001000000]: dynamic reconfigure is not
enabled for this image topic [raspicam_node/image] becuase <cameraName> is
not specified

so I think the markers images aren't being spawned:

  Publish frame odom on tf: enabled
Warning [parser_urdf.cc:1137] multiple inconsistent <gravity> exists due to
fixed joint reduction overwriting previous value [false] with [true].

[ INFO] [1631214090.998847909, 0.087000000]: Physics dynamic reconfigure
ready.
[ INFO] [1631214091.056776896, 0.100000000]: Updating map with 0
observations. Map has 16 fiducials
[fiducial_marker_spawn-16] process has finished cleanly
log file:
/home/dr1bot/.ros/log/4a466a56-11a0-11ec-bb02-3f3f629dbd6c/fiducial_marker_spawn-16*.log
[ INFO] [1631214093.040701289, 0.500000000]: Updating map with 0
observations. Map has 16 fiducials

anfederman, 2021-09-10:
OK here's the breakdown

magni Gazebo fiducial world launch display fiducial grid latest bugfix
branch do the fiducials display:

Noetic acer laptop  -  no
Raspberry PI ARM64 noetic - no
Jetson Nano melodic -   yes

I will try on another laptop/ AMD/Intel platform, but I am losing interest
in trying to track this glitch down.  On the Acer it is saying not
observing markers on the map, on the Pi it is complaining about gravity.

So for now let's close the issue, since it seems it will work on other
architectures.

It should be pulled from our regular magni_robot distro - since it isn't
supposed to have gazebo on it.

JanezCim, 2021-09-13:
If this is really a platform issue someone else should try to test this if it works. @JanJericevic can you try launching `roslaunch magni_gazebo fiducial_world.launch` on the` bugfix-gazebo-extrinsicsfolder` branch if it works for you?

JanezCim, 2021-09-20:
@JanLipovsek tried this out and it worked on Noetic. So how I would go about this is that if you have a specific platform that this is not working on, you should open a separate issue for that thing specifically and then we can solve it there. For now i'd call this solved

closed 2021-09-20 by JanezCim

anfederman, 2021-09-21:
FIducial world now launched on all four of my tested platforms.