UbiquityRobotics/magni_robot #175: magni_description missing install rules

rohbotics, 2021-09-06:
We need to add the cmake install rules for the default extrinsics, they are not present in the debian version of the package.

JanezCim, 2021-09-07:
When making the current state I was planing that the default extrinsics would live in `magni_robot/magni_description/extrinsics/`. So im not sure what you mean with "add the cmake install rules for the default extrinsics"

anfederman, 2021-09-07:
Except the Magni gazebo launch doesn't find the camera in noetic because
the extrinsics folder isn't found. On the beta, the extrinsics folder was
missing, or perhaps under a param folder?

anfederman, 2021-09-07:
This is still an issue in that in magni_gazebo  it was still looking for the extrinsics file in param  which didn't exist in /opt/ros/noetic/magni_description.    I fixed it locally by "sudo cp -r extrinsics param"

JanezCim, 2021-09-08:
hmm, that is strange because it works on my Ubuntu 20.04, ROS Noetic machine just fine: 
by running 

`roslaunch magni_gazebo fiducial_world.launch`

[image]

What machine are you using? Can you recheck that you are checkout and pulled the branch `bugfix-gazebo-extrinsicsfolder`? What command are you using to run the gazebo?

anfederman, 2021-09-08:
***@***.***:~/catkin_ws/src/magni_robot$ git branch
  bugfix-gazebo-extrinsicsfolder
* noetic-devel
***@***.***:~/catkin_ws/src/magni_robot$

 [fiducial_marker_spawn-16] process has finished cleanly
log file:
/home/dr1bot/.ros/log/9120ef1c-10c6-11ec-ad8a-1778231a9cf4/fiducial_marker_spawn-16*.log
[ INFO] [1631120616.895179499, 0.700000000]: Updating map with 0
observations. Map has 16 fiducials
[ INFO] [1631120617.555781574, 0.800000000]: Updating map with 0
observations. Map has 16

but I only see the bottom fiducial. I'll try on the ARM64 pi and see if
there is any difference. Also, on RVIZ,I don't see the robot model until I
change from odm to base_link.

JanezCim, 2021-09-09:
> ***@***.***:~/catkin_ws/src/magni_robot$ git branch
>   bugfix-gazebo-extrinsicsfolder
> * noetic-devel

This seems to be indicating that you are on `noetic-devel` branch, You should be on `bugfix-gazebo-extrinsicsfolder` branch. Please do so with 

```
cd ~/catkin_ws/src/magni_robot/
git checkout bugfix-gazebo-extrinsicsfolder 
git pull
```

recompile and try again

rohbotics, 2021-09-09:
Hey  @JanezCim  

The issue is that by default only "installed" components of the package will be included in a binary distribution, and currently we are not installing the extrinsics. They need to be installed just like the launch and the urdf directories here:  https://github.com/UbiquityRobotics/magni_robot/blob/noetic-devel/magni_description/CMakeLists.txt#L114

JanezCim, 2021-09-10:
@rohbotics oh okay i think i see the problem, but then I suspect that @anfederman is experiencing a completely different set of problems.

I've included the install rules in #177 

@anfederman  i think this is a discussion that is a duplication of #174 anyway, so lets continue solving that there.

anfederman, 2021-09-10:
On my laptop, I might be missing something. For instance I have Magni robot
and fiducials, but no Raspicam node. So even though there is an extrinsics,
something else is missing. Same thing might be true on the ARM64 RPi., But
Raspicam node is there. It might be missing some dependacy.

closed 2021-09-21 by rohbotics

anfederman, 2021-09-21:
BTW  the bugfix has eliminated the magni_gazebo fiducial world not
displaying fiducials issue, which occurred on some of my platforms.

JanezCim, 2021-09-22:
@anfederman ok we can continue the discussion for fidducials issue on #174

PR #179 (2021-09-21): Added install rules for extrinsics
Fixes #175