UbiquityRobotics/magni_robot #180: tracking bug-fix issues

anfederman, 2021-09-22:
Trying to figure out which branch should be used for testing is confusing, especially when the issue being addressed isn't specified.
For example the issue called bugfix-extrinsic got changed to bugfix-175.  I no longer know what should be pulled/checkout when testing different parts of code.

JanezCim, 2021-09-23:
I think you dont have to test every branch that exsists in the repo (some of them are stale or only for testing). What should rather be tested are the PRs and their descriptions should include the instructions for testing that. Naming a branch bugfix-175 just indicates that this branch is trying to fix issue "#175" of the same repo

JanezCim, 2021-10-01:
@anfederman hopefully this is clear now? If not please reopen issue

closed 2021-10-01 by JanezCim