UbiquityRobotics/magni_robot #184: Change Lidar location on Top plate

JanezCim, 2021-10-01:
Zdej pa šeenkrat spremenjene želje ker so na sestanku včeraj ljudje želeli malo drugačno postavitev ekrana in lidarja na magni-ju:

dodatne info:
 - ekran je centralno poravnan
 - za inštalacijo ekrana bomo uporabili m3 standoffe tako da bi bile lokacije lukenj malo drugačne - na sliki
 - lidar luknje ostanejo isto razporejene med seboj, samo lidar lokacija bi se spremenila, tako da pride malo bolj nazaj (tako da lidar ne ovira vijakov za prišravfat top plate na robota) in malo na stran (tako da če hočjo ljudje na center kaj monterat da ševedno lahko)
 - ko se odločiš za natančno lokacijo lidarja nam prosim povej mere x, y na sliki magnicable

Rabiš še kaj informacij?

AljazKrsnik, 2021-10-04:
Hello

JanezCim, 2021-10-06:
Hey Aljaž how is this going?  :)

AljazKrsnik, 2021-10-06:
[image]

AljazKrsnik, 2021-10-11:
Posiljam top plate brez izpranitve. Mislm, da je pozicija liderja dobro
definirana v x in y. Če pa rabiš dodatno kaj oz. nisem razumel pa povej.
Lp Aljaž

AljazKrsnik, 2021-10-13:
> Okay I see it now, thats ok, can you also attach the 3d file without the
> square hole and then we can close this :
> [image: wohole]
> <https://user-images.githubusercontent.com/8426792/136910648-1e2e11d4-5e85-46ed-a0d3-25b565da928c.png>
> )

File ki sem ti ga poslal, je že brez te luknje. Lp

JanezCim, 2021-11-05:
True, @AljazKrsnik on your measures pdf i could not find these measures, can you provide them?
[image]

JanezCim, 2022-01-31:
for future reference on last image x: -0.092m, y: -0.066m

JanezCim, 2022-02-02:
@AljazKrsnik this model has been tested like described above and i think thats fine. If its part of our default magni 3d model, then this is solved.