UbiquityRobotics/magni_robot #186: Lidar temporary mechanical mount extention

JanezCim, 2021-11-02:
We need to make a LSlidar mount extention board that will be a temporary solution before top plates get into production with standard magni. Confinements:
 - small board that will mount the LS lidar onto the magni base robot
 - small enough to fit into the box with lidar - with which it will be shipped
 - ideally lidar would be in the same side position as it is on the new top plate

JanezCim, 2021-11-03:
The inside edge of the box the Lidar is packed in is 165mmx175mm. There's a gap between the lid and the stirofoam layer of 8mm.

AljazKrsnik, 2021-11-04:
Can you send me/show me the shape of the top plate that they have already made 100 pcs?

JanezCim, 2021-11-05:
the top plate that they already made is the standard top plate that was a part of the robot until now. I cant find a picture or a 3d model of it anywhere right now. 

Its the same top toplate that you were modifying here https://github.com/UbiquityRobotics/magni_robot/issues/184 but without the lidar, cable or the screen holes

AljazKrsnik, 2021-11-05:
Ok. Thx

AljazKrsnik, 2021-11-14:
[image]
[image]

Here is a mount. If i understood you correctly there is only 8 mm of free space in height?
With this mount, we are gonna have same lider position( latest). I can also make aditional holes for the new lidar?
We must ask Petal pečnik if he can make thak kind of part becouse of small amount of material on the curwin point.

JanezCim, 2021-11-15:
@AljazKrsnik if I understand correctly that green mount is made to fit onto the Top Plate (gray) right? This mount is meant as a replacement for the top plate, and does not actually need to sorround it. So i would maybe rather do it as a simple plate (shape is already nice) - without any bending parts. Maybe it'll be quicker if we discuss this on a call

AljazKrsnik, 2021-11-15:
I see. That is a bit different. So we make a holder like this? Or if we really need stifness becouse of the LIDAR, we can make 2 mm thick sheet metal and i think that it would be suficient? Dont know the weight of the lidar...
For the mounting we can chanfer the holes( than we must go with al least 2 mm), or we can bolt it and use other emtpy holes.
[image]
[image]

JanezCim, 2021-11-17:
Yes that looks better then before. Do you think chanfering is absolutely needed since it adds complexity to the production process?

I also dont know the weight of the lidar sory. But you can also make this from 3mm aluminium since that is how thick the top plate is anyway.

Just one more thing to note, after we have made some test units of this plate for us at Pečniks place and tested them, we will try to produce the rest of them at our production contact in China - since these need to be shipped to our lidar suplier which is also in china. This is so we try to avoid tarifs.

AljazKrsnik, 2021-11-17:
No problem. I will make it without chanfering.

JanezCim, 2021-12-15:
@AljazKrsnik  Is this part ready to be produced then after we've tested it in TP? If yes, can you point me to the correct file on Zoho?

JanezCim, 2021-12-15:
Oh and also on the test units in TP dont have these holes. That may be good to be added to that design.

[image]

JanezCim, 2021-12-21:
@AljazKrsnik can you send a zoho link to the file which i can send to production for this temporary lidar mount if its ready for production?

AljazKrsnik, 2021-12-21:
Will send today( evening hours).

AljazKrsnik, 2021-12-24:
Here is the new, updated mount with aditional holes.
https://workdrive.zohoexternal.com/external/2c910607cb649df274d970fb7bda6b4dd066c89eb6d7f333686d835c7470af84

closed 2022-01-10 by MoffKalast

JanezCim, 2022-03-16:
temporary mounts from Leishen are in slovenia, now we need to test them:
 - is colored blue RGB(48, 94, 177) - is it colored or annodized?
 - Each sample unit should come with 4xM3x8mm Cap socket screws.  - is this true?
 - mount on robot and mount ls lidar on it - does everything fit?
 - mount ld lidar on it - does it fit? 
 - how many mounts have arrived - should be 10?

@GreTimotej once you test this please message this to Neja and she is going to organise sending of mounts to customers. Also message here if all test are good so i can message stephen that we have a green light for more.

GreTimotej, 2022-03-17:
After today's tests:
- The piece is colored blue, but a bit darker than RGB(48, 94, 177). At least that is how it looks compared to the color viewed on the computer. It is anodized, ([See pictures here](https://workdrive.zohoexternal.com/folder/qc97hc5bc2475baac4825bf6137adce9526ca)).
- Alongside the 10 mounts we also got 40 M3x8mm Cap socket screws. That is 4 screws per piece. They were also measured and I can confirm that they are 8mm long.
- The mount can be mount on the robot, but one of the holes is a bit wrong. The screw can not be put in all the four holes, because of the tension against one side([see picture](https://workdrive.zohoexternal.com/folder/qc97he347ff9d83084fda85976063fd102145), how the screw has less space on right side of the hole than on the left side). The same if with all the other holes.
- LS Lidar can be mount - it fits good. One small thing I noticed is that the screw thread is a bit poorly cut and the screws are sometimes a bit hard to screw into the thread. ([see pictures](https://workdrive.zohoexternal.com/folder/qc97hed2bfc65074e41ada23cea8dad387867)) here
- LD Lidar does fit, but the screw thread issue applies here too. ([see pictures here](https://workdrive.zohoexternal.com/folder/qc97hbca8d0b8af7c4504b3ef5464ffd9ece5))

AljazKrsnik, 2022-03-17:
I will look the 3d files if the holes are concentric or not and report
today( majbe 3d file wrong or press in nuts inserted at an angle)
Is the lidar stable with just 2 screws( as planed)?
Does the lidar( sheet metal) oscilate or shake when MAGNI moves?

GreTimotej, 2022-03-17:
@AljazKrsnik Yes the Lidar is stable with just 2 screws and when Magni is moving everything is still (Mount mounted with 2 screws and LS Lidar also mounted with 2 screws).

AljazKrsnik, 2022-03-20:
Hi. The MAGNI 5 has only 3 inserts on FRONT SUPPORT( number 6 -https://workdrive.zoho.com/home/g9vop387248f4f5304dde876eb982824d88c5/teams/g9vop387248f4f5304dde876eb982824d88c5/ws/g9vop0ead768c4254422e9332d4e4fd4a414b/folders/ktgta31e7040cc0d64d23a41dfc5f4eac3ead). Look the link.

Here is another picture for this LIDAR mount.
[image]
I looked my files and the holes are coaksial. I would look just the holes made in the sheet metal, and check if they are really coaksial( if they have the latest files). Or the inserts are not coaksial inserted. 

When putting in the screws, do NOT fully tighten them. Screw all in but fully tighten them when all are almost fully tightened.

anfederman, 2022-03-20:
My top plate has the holes for top center.

JanezCim, 2022-03-21:
@anfederman yes, you got top plate that was ment only for testing. The top plates and temporary plates in all the future builds will have the lidar holes slightly offset to robots right

JanezCim, 2022-03-25:
@GreTimotej I have to answer to the producer of these holders, what is the status now of this? Are the holes coaxial then and can i give a green light for producing more holders OR do we have to do another iteration?

GreTimotej, 2022-03-25:
@JanezCim yes, sorry. The holes are coaxial and you can give the green light for the production.

JanezCim, 2022-03-25:
Great, thanks!

anfederman, 2022-10-11:
Ok does that mean we need to modify the laser filter yaml and static
transform for various configurationd?

AljazKrsnik, 2022-10-11:
Will the small lidar mounting holes need adjustment?