UbiquityRobotics/magni_robot #192: Styrophome Packaging 3D Standard work instructions

JanezCim, 2022-01-24:
@AljazKrsnik I've got 3D files for **new syrophome packaging**: https://workdrive.zoho.com/folder/8x05cf73c06f4f97648cb94343140f18889a3

What we need to do it update the magni standard work instructions with the new packaging: https://writer.zoho.com/writer/open/456pt0c2a3e68e60e4804b0d4b084b3294656

Specifically, import these into solidworks, insert magni, create screenshots, insert screenshots into standard work

AljazKrsnik, 2022-01-24:
Cannot open. Can files be saved in step file? I cannot open them.

MoffKalast, 2022-01-26:
[image]

AljazKrsnik, 2022-01-26:
:D

JanezCim, 2022-01-26:
@AljazKrsnik I will ask the guy in the mail and include you

AljazKrsnik, 2022-01-27:
I am working on package assembly. But i am a bit confused. I do not know where to use top sides? It seems that the whole package assembly is kinda ok without 
[image]
[image]
[image]
[image]
"top sides" part.

JanezCim, 2022-01-28:
yeah looks wierd. Can you ask Vasyl on mail about it?

JanezCim, 2022-01-28:
Btw look at magni standard work step 71. I think they Vasyl forgot to add a piece of styrophome that would fit here 
[image]

Do you see that also from the 3d model?

For current units the manufacturer is creating the missing piece from bubblewrap and there have been complains about the part missing. When you send a mail to Vasyl, be sure to include me so i can tell him this.

AljazKrsnik, 2022-01-28:
Ok. Will ask him for more info.

JanezCim, 2022-02-01:
Any progress on this?

AljazKrsnik, 2022-02-01:
Yes. Will do this first. Got the mail from vasily, how to pack all together.

AljazKrsnik, 2022-02-07:
HI

I have made 3D catalogue for styrophome packaging. Tell me what you think. 
[MAGNI 5 WITH PACKAGE.PDF](https://github.com/UbiquityRobotics/magni_robot/files/8015110/MAGNI.5.WITH.PACKAGE.PDF)

I have renamed some ofthe packaging files( more user friendly)

JanezCim, 2022-02-07:
Great! thanks @AljazKrsnik i've added the file to the standard work. If it does not present too much work for you, could you also recreate steps 68-83 with rendered images and try to replace them in the magni standard work:
https://writer.zoho.com/writer/open/456pt0c2a3e68e60e4804b0d4b084b3294656

AljazKrsnik, 2022-02-09:
[image]

What is here missing?

AljazKrsnik, 2022-02-09:
[image]

Is this FOAM  insert still present? Do i include it in assembly mauna?

AljazKrsnik, 2022-02-09:
https://writer.zoho.com/writer/open/456pt0c2a3e68e60e4804b0d4b084b3294656

I have created new version. There are instructions for the new packaging system. The only thin i am not sure is about INSER FOAM on the upper picture. Will this still be a part of the new packaging assembly?

JanezCim, 2022-02-10:
> What is here missing?

This: https://github.com/UbiquityRobotics/ubiquity_main/issues/255 david needs to arrange with lee pulver that we get updated certificates of conformance.

> Is this FOAM insert still present? Do i include it in assembly mauna?

I think yes, its still present

JanezCim, 2022-04-08:
@AljazKrsnik the syrophome packaging for this is in standard work already and done righ? If yes, you can close this issue with the button underneath :)

closed 2022-04-12 by AljazKrsnik