UbiquityRobotics/magni_robot #195: description.launch  undocumented

anfederman, 2022-03-17:
I could find no documentation on how to specify that I had sonars, OLED, lidar, etc.   ON a fresh launch I am not sure if the etc/ubiquity/robot.yaml is being read.  Since the default_robot.yaml is empty This is confusing.

PS the MArch distro had magni_robot in catkin,   put raspicam, aruco, pi_sonar  in /opt/ros   I am not sure what happens if you do a
fresh install via apt.  I expect issues.

<launch> 
  <arg name="sonars_installed" default="true"/>    ???????????????????
  <arg name="camera_extrinsics_file" default=""/>  ??????????????????
  <arg name="lidar_extrinsics_file" default=""/>   ???????????????????
 
  <param name="robot_description" 
      command="$(find xacro)/xacro --inorder
        '$(find magni_description)/urdf/magni.urdf.xacro' 
        sonars_installed:=$(arg sonars_installed)
        lidar_extrinsics_file:=$(arg lidar_extrinsics_file)
        camera_extrinsics_file:=$(arg camera_extrinsics_file)
        "
  />
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_sta
te_publisher" />
</launch>

anfederman, 2022-03-17:
<launch>
  <arg name="sonars_installed" default="true"/>
  <arg name="camera_extrinsics_file" default=""/>
  <arg name="lidar_extrinsics_file" default=""/>

  <param name="robot_description" 
      command="$(find xacro)/xacro --inorder
        '$(find magni_description)/urdf/magni.urdf.xacro' 
        sonars_installed:=$(arg sonars_installed)
        lidar_extrinsics_file:=$(arg lidar_extrinsics_file)
        camera_extrinsics_file:=$(arg camera_extrinsics_file)
        "
  />
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_sta
te_publisher" />
</launch>

anfederman, 2022-03-17:
for some reason, when I try to paste the desciption.launch file, it disappears.

JanezCim, 2022-03-18:
Which image EXACTLY are you on (please give full name)? - saying march image is not enough since there were more then 1 march image
Is the image fresh? If not can you re-install and try again?

> default_robot.yaml is empty 

this should not be empty, please make sure you are on fresh image without your mods.

> I could find no documentation on how to specify that I had sonars, OLED, lidar, etc. 

You are right, there is nothing documented in READMEs about this, but there is comments inside default_robot.yaml that should give users some clue how to specify weather sonars, camera and lidar are enabled, disables and their extrinsics. Is that not enough? If no, where in the documentation would you suggest putting this that it would be the most seen by users?
https://github.com/UbiquityRobotics/magni_robot/blob/c8a130c8caf8ba3b26974da581bb97b2957da20b/magni_bringup/config/default_robot.yaml#L14

> ??????????????????? ?????????????????? ??????????????????? 

Seems like you have a bit of keyboard troubles with your questionmark key being a bit stuck? :D

anfederman, 2022-03-18:
By empty I mean it just has comments, and no directives. So I edited etc
ubiquity robot.yaml to turn sonar on. The image is the 3-10 to the best of
my recall. I am away from my desk.

MoffKalast, 2022-03-18:
> for some reason, when I try to paste the desciption.launch file, it disappears.

Have you tried calling the Ghostbusters?

anfederman, 2022-03-19:
Ok. My symptoms now are pi_sonar isn't starting from boot up. I can launch
it, and it echoes topics, when I try to see the sonar, in rviz, the cones
don't appear and it gives a Tf2 frame error. I can launch aruco, it appears
to work. I can launch LSlidar, but it doesn't show up in rviz, it looks
like a static transform error. my image is 03-04-ubiquity-gdm3-without-oled.

JanezCim, 2022-03-21:
> for some reason, when I try to paste the desciption.launch file, it disappears.

Thats because youve copy pasted the xml code into it without marking it as a code snippet with ``. Github actually renders the xml code if you dont format it with code. i fixed it now in your comment, please see what has changed.

JanezCim, 2022-03-21:
> Ok. My symptoms now are pi_sonar isn't starting from boot up.

Thats because in `/etc/ubiquity/robot.yaml` it says

`sonars_installed: False `

Put it on True and they should work. Afaik, in Kinetic we had sonars disabled by default. So its the same in noetic.

JanezCim, 2022-04-08:
closing because of inactivity, reopen if needed

closed 2022-04-08 by JanezCim