UbiquityRobotics/magni_robot #200: lidar detects robot in gazebo

JanezCim, 2022-03-23:
running `roslaunch magni_gazebo empty_world.launch`

[image]

JanezCim, 2022-03-23:
found out that commenting out sensor-hc-sr04 sensor dae files solves the problem 
https://github.com/UbiquityRobotics/magni_robot/blob/cfb8f85852e54119f8eda4ff8be8f6d91a0341cf/magni_description/urdf/sensors/sonar_hc-sr04.xacro#L18

so probably the dae file is the problem, wed have to change it. Anyway, a quickfix for this if you dont need sonars is to set the sonars to false

https://github.com/UbiquityRobotics/magni_robot/blob/cfb8f85852e54119f8eda4ff8be8f6d91a0341cf/magni_gazebo/launch/empty_world.launch#L17

MoffKalast, 2022-03-23:
Hmm right, I think we had this issue before, but I can't recall what the fix was (or if there was any). There's probably an issue logged somewhere about it.

JanezCim, 2022-03-25:
yeah i remember this as well but could not find it...

closed 2022-04-01 by MoffKalast