UbiquityRobotics/magni_robot #207: Magnigazebo/launch/rviz_config.rviz

anfederman, 2022-06-10:
The current rviz_config.rviz in magni_gazebo launch doesn't parse the urdf correctly. The breadcrumb_gazebo/launch/rviz_config.rviz
is correct, so if someone would be so kind to test this and copy the file over, I would appreciate it.

MoffKalast, 2022-06-10:
The difference is that the gazebo one also launches magni_description, which solves the problem.

MoffKalast, 2022-06-28:
Well we should find a way to solve this eventually, right now my workaround is to just launch magni_description separately which does work but also publishes duplicate transforms which causes other problems. We probably just need to add the model publisher or something, I haven't really looked into it.

anfederman, 2022-07-20:
I got it working.

closed 2022-07-20 by anfederman