UbiquityRobotics/magni_robot #220: transform for camera forward displays incorrrectly in RVIZ

anfederman, 2023-01-18:
https://drive.google.com/file/d/13rZr6WvJyQTN6nPpjQd4De3GtmNIorb4/view?usp=share_link

This video show the issue in rviz.

You also can demonstrate by selecting camera forward in magni_bringup (default.yaml) and running:

roslaunch magni_gazebo fiducial_world.launch

When you do this you will find the fiducials projected beneath the floor in rviz.

In a real robot situation,  left and right for marker positions are flipped from the real world. 

I don't understand why the urdf and frame transforms aren't working correctly.  This doesn't seem to be an issue for other extrinsic files.

anfederman, 2023-02-02:
Apparently none of the extrinsics files are being parsed correctly on my magni_silver.  No matter what I specify for the camera or lidar, the camera is in the forward xyz position with the rpy being ignored, the lidar is in top_plate(right), and won't relocate to top_plate_center.

This doesn't happen in magni_gazebo on my laptop.

This is on noetic_devel on a fresh download of magni_robot built on a Pi 3.

Can some one please test this on a Pi 3 and see if this is an isolated problem only I am encountering?

BTW the lidar is working, the camera is working, the robot drives with the logitech, and other then fiducials generating quaternion errors, no obvious errors or warnings.

MoffKalast, 2023-02-02:
> Can some one please test this on a Pi 3

Which museum do we get it from?

anfederman, 2023-02-02:
Ha ha you should try some stand-up comedy.

anfederman, 2023-02-03:
source of issue was:

Skipping time sync steps due to configuration
Camera enabled with extrinsics: /home/ubuntu/catkin_ws/src/magni_robot/magni_description/extrinsics/camera_extrinsics_ahead.yaml
Lidar enabled with extrinsics:: /home/ubuntu/catkin_ws/src/magni_robot/magni_description/extrinsics/lidar_extrinsics_top_plate.yaml

for some reason the camera position being loaded was for ahead, not forward.

by editing the camera_estrinsics ahead to be correct for the actual camera pose on my robot, the problem was resolved.

I am still uncertain where the camera ahead load is being picked up, but I suspect it is in aruco.launch.

anfederman, 2023-02-03:
Skipping time sync steps due to configuration
Camera enabled with extrinsics: /home/ubuntu/catkin_ws/src/magni_robot/magni_description/extrinsics/camera_extrinsics_ahead.yaml
Lidar enabled with extrinsics:: /home/ubuntu/catkin_ws/src/magni_robot/magni_description/extrinsics/lidar_extrinsics_top_plate.yaml
 
bye editing camera ahead, problem resolved. uncertain where this is being picked up, expect in aruco.launch